function3d/petalot #2: Always false condition in hotend.hpp

javiersanzsanchez, 2022-10-27:
On strat method the condition "if (tempLastStart == 0)" will never be true do to "tempLastStart = millis();" on previous line.

[image]

function3d, 2022-10-27:
In the "stop" method there is a "tempLastStart=0" so after a manual or end filament "Stop" event, "if (tempLastStart == 0)" will be true

javiersanzsanchez, 2022-10-27:
I think no... you set "tempLastStart = millis();" so it will never be zero after that.

function3d, 2022-10-27:
the code of the start method is only executed the first time it is called after a stop, the following times it is called it is not executed because of that condition

function3d, 2022-10-27:
anyway, what issue is this condition causing you?

javiersanzsanchez, 2022-12-05:
No, is not completed

closed 2022-12-05 by function3d

function3d, 2022-12-06:
I'm not going to recode something that works no matter how amateurish you think it is