function3d/petalot #3: Probleme pcb

Stevoidex, 2022-11-04:
Bonjour je trouve Fantastique votre travail j ai voulu reproduire ce que vous avais fait tout fonctionne jusqu au moment de toute allumer la carte D1 ce mes a bruler sur le composant u3 je les refait 2 fois et j en ai bruler 2 j'utilise une alimentation 12v 5a si vous pouvez m aider ça serai sympa

Stevoidex, 2022-11-04:
[image]

Stevoidex, 2022-11-04:
[image]

Stevoidex, 2022-11-04:
[image]

Stevoidex, 2022-11-04:
https://zupimages.net/up/22/44/nou6.jpg

Stevoidex, 2022-11-04:
[image]
[image]

Stevoidex, 2022-11-04:
[image]

function3d, 2022-11-04:
Did you set the U3(HW-613) to 5 volts with the potentiometer?
Avez-vous réglé le U3(HW-613) sur 5 volts avec le potentiomètre?

Stevoidex, 2022-11-04:
Merci pour votre réponse, Non  c'est ce que j ai vue en cherchent mes je ne connais pas du tout  que faut t il souder ? Le 12v et le 2. 5v ? Faut t il donner un coup de couteau dans la ligne noir ? Comment fonctionne le potentiomètre

function3d, 2022-11-04:
You put 12v between IN and GND, and with a multimeter you measure VO and
GND, turn the potentiometer until you measure 5V, another option is to cut
the jumper on ADJ and put a drop of tin on the 5V jumper

function3d, 2022-11-05:
Vous mettez 12v entre IN et GND, et avec un multimètre vous mesurez VO et
GND, tournez le potentiomètre jusqu'à ce que vous mesuriez 5V, une autre option consiste à couper
couper le cavalier sur ADJ et mettre une goutte d'étain sur le cavalier 5V.

Stevoidex, 2022-11-05:
Merci beaucoup j ai tournez le potentiomètre sur 5v comme vous l avais expliqué c est nickel  la buse chauffe le détecteur filaments fonctionne mes le moteur ne tourne pas je les tester il fonctionne mes y a t il quelque chose a tourné sur le la carte moteur pas à pas ou a t elle pu griller comment la tester

function3d, 2022-11-05:
The stepper motor driver has only one valid position assuming you have it properly plugged in, if it does not work it is probably broken, the only way to know if it is broken is to try another driver

function3d, 2022-11-05:
Never connect/disconnect the motor with the electronic on

https://github.com/function3d/petalot/blob/master/before%20you%20switching%20on%20the%20electronics.md

Stevoidex, 2022-11-05:
J ai meusurer et il meusure bien 0,4 v le moteur ne tourne pas je vais donc voir pour en racheter une merci encore pour votre aide

Stevoidex, 2022-11-14:
Bonjour j en ai racheter un nouveau tout fonctionne mes le problème maintenant est le moteur ne tourne pas vraiment il vibre

https://user-images.githubusercontent.com/117478768/201705312-42afa9d7-dcf4-4057-8cb5-780f0ce6632e.mp4

function3d, 2022-11-14:
No, ignore my previous comment, I checked and the cables are interchanged on my machine as well.

function3d, 2022-11-14:
The stepper motor has two windings, the vibration can be caused by incorrect wiring and also by an unpowered or fused winding. To find out for sure you need to try with another operating stepper motor if it does the same we rule out that the engine is damaged, or connect this motor to the extruder cable of your 3d printer and spin it extruding in your printer.

Stevoidex, 2022-11-14:
Merci beaucoup pour votre aide je vien de trouver c est le moteur pas a pas qui n ai pas bien enfoncé dans les pin

closed 2022-12-05 by function3d