function3d/petalot #6: Cannot connect

NickGreen98, 2022-12-04:
I cannot connect to the D1 mini to set up the wifi. I have tried multiple times. adding a way to manually enter the wifi info before flashing would be good.

function3d, 2022-12-18:
When you upload the firmware for the first time, the Wemos D1 starts a WIFI access point called PETALOT, you connect to this WIFI and navigate to the ip 192.168.1.4, here you put a free ip in your netword for the Wemos D1 as well as the parameters of your local wifi, your SSID, your WIFI password, your gateway and 255. 255.255.0 in the subnet mask, submit the form and if all goes well the Wemos D1 will connect to your WIFI and you will be able to access it through the ip you assigned in the previous step.

NickGreen98, 2022-12-18:
The WeMos D1 broadcasted the local network called petalot. None of my
devices would connect to the network to fill out the form.

function3d, 2023-01-02:
working on a version that works directly by loading the firmware without the need to enter wifi data and with the possibility to configure the device from the terminal via serial communication.

function3d, 2023-01-02:
The new version is uploaded, if you decide to try it, to configure the device you can use the serial monitor of arduino IDE, after uploading the firmware, open arduino IDE and its serial monitor and restart the wemos, in the serial monitor you will see a JSON with the configuration, copy and edit it complete and paste it complete, once edited, in the serial monitor input, press enter, the device will use your edited JSON input as new configuration and will restart so it will start the process of trying to connect to your WIFI if you have entered the WIFI/network data in the new configuration.

closed 2023-01-02 by function3d

NickGreen98, 2023-01-04:
Hello,

I am still not able to connect. I am able to upload the binary using esphome-flasher, but I am still unable to upload using arduino IDE. It is missing the FS.h header file mentioned in conf.hpp, and I don't know what library that came from. 

As for using the serial monitor to get the ssid and pass in, here is my output, I took out the ssid and password for safety: 

`
To:Temperature
Vo:Speed
Tm:Maximum Temperature
Kp:Kp
Ki:Ki
Kd:Kd
R1:R1
Max:Maximum value for MOSFET (0-255)
ssid:SSID
password:SSID Password
LocalIP:IP address
Subnet:Subnet
Gateway:Gateway
ifttt_event_name:IFTTT Event Name
ifttt_api_key:IFTTT API Key
{"To":220,"Vo":40,"Tm":250,"Kp":23,"Ki":0.043,"Kd":160,"Max":210,"ssid":"SSID","password":"PASSWORD","LocalIP":"192.168.0.100","Subnet":"255.255.255.0","Gateway":"0.0.0.0","R1":10000,"ifttt_event_name":"","ifttt_api_key":""}
...................AP Ready
`

function3d, 2023-01-11:
Hi,

FS.h is part of the core of Arduino ESP framework, did you installed it ?
https://github.com/esp8266/Arduino

"ssid":"YourWifi","password":"YourWifiPassword","LocalIP"
:"192.168.0.100","Subnet":"255.255.255.0","Gateway":"YourIpGateway"
These parameters are from your current WIFI, with these parameters the
electronics works as a client of your WIFI network instead of as an access
point.

Regards
Raul,  function.3d