function3d/petalot #12: Negative temperature value in UI

FUF1, 2022-12-25:
Thanks for making your amazing project publicly available.

I built it based on your provided schematic and used a [100k thermistor](https://3d-druckershop.com/p/thermistor-ntc-3950-100k-ohm-kaufen). Weirdly enough I get an unplausible temperature value in the UI with around -273°C (which is 0 Kelvin - so probably the measured value is unavailable for the calculation?). Of course, switching the thermistor leads does not change anything. Changing the R1 value in the settings does change the readings but not in a fruitful way. The heating stops at around 100°C (measured directly with an external probe at the nozzle).

I tried to figure out the causation of this by reading your code. But I must say, it is very advanced and I can't pin down the problem, if it is hardware or software related. I thought you or someone else here might have experience with a similar issue through your own work. Thanks in advance.

function3d, 2023-01-02:
What voltage do you supply to the thermistor?

FUF1, 2023-01-02:
Hey, thanks for your reply. I used the 3.3V noted in the schematic. After reading through different forums I found a solution that seems to work, altough I don't know if it is the standard way to do things: I put the thermistor to my 3d printer hotend to check the resistance at the target temperature and then used similar values (they were much lower, around 500-800 ohms, because the resistance drops massively with this thermistor when heated up). Then I just fired up petalot and measured the nozzle temp externally and swapped resistors until it was somewhat correct (deviation of around 3°C now). This is good enough and works. Not the problem is that the steppermotor has not enough torque to pull the filament through, but I will switch to another motor and a bigger gear train. Thanks for all your work!

closed 2023-01-02 by FUF1

function3d, 2023-01-02:
The drop in thermistor resistance is normal, this is how thermistors work, the variations in resistance is how we know what the temperature is.

The motor of my system is enough to make the filament, probably you need to cut the strip narrower, each type of bottle has its thickness and therefore its width, https://function3d.xyz/en/?Display_FAQ=267

FUF1, 2023-01-03:
Thank you for your answer. That is a good hint. I'll try to make it even smaller and experiment with it based on your list in the link.