function3d/petalot #15: Stepper not spinning

McCakeAh, 2023-01-24:
We set up all of the components just as you had done and all worked except the stepper motor just made noise and vibrated. Any suggestions on what might be causing this? Could pins on the stepper be in the wrong order, or a bad stepper? 
Thanks

function3d, 2023-01-24:
https://github.com/function3d/petalot/blob/5b1d409dda7f66e7040381943212c45b0bf8b62b/before%20you%20switching%20on%20the%20electronics.md#before-switching-on-the-electronics-for-the-first-time

McCakeAh, 2023-01-24:
> https://github.com/function3d/petalot/blob/5b1d409dda7f66e7040381943212c45b0bf8b62b/before%20you%20switching%20on%20the%20electronics.md#before-switching-on-the-electronics-for-the-first-time El mar, 24 ene 2023, 23:22, McCakeAh ***@***.***> escribió:
> […](#)

we have tried all that and still the same results

function3d, 2023-01-24:
Check the connector, it may be loose

function3d, 2023-04-04:
Check your welds, make sure you do not have two adjacent welds joined together.

closed 2023-04-12 by function3d