function3d/petalot #21: Stepper motor doesnt work

MarmotLand, 2023-03-15:
A recently set up everything and it works but accept for stepper motor I set up the the voltage 0.4 V the stepper motor doesnt do anything any noise any spin nothing THANKS for help

gulsinho06, 2023-04-02:
Did you buy the stepper motor indicated in the video and another? if it is another one, make sure its vref and readjust the driver voltage

MarmotLand, 2023-04-02:
> Did you buy the stepper motor indicated in the video and another? if it is another one, make sure its vref and readjust the driver voltage

Im using NEMA 17 17HS3401 and i reacently came up to a conclusion that the stepper motor driver might be burnt , because I accidently disconnected the steper motor while i was testing it. And one more question whats vref ? Thanks for replying

gulsinho06, 2023-04-02:
You're welcome

function3d, 2023-04-04:
Check your welds, make sure you do not have two adjacent welds joined together

closed 2023-04-12 by function3d