function3d/petalot #26: Arduino does not create wifi network

RuhtraXD, 2023-04-01:
I followed every step by step but the arduino does not create the wifi network called petalot.

gulsinho06, 2023-04-02:
Instale o driver do esp8266 para o chip CH340G e compile novamente

gulsinho06, 2023-04-02:
Install esp8266 driver for CH340G chip and compile again

RuhtraXD, 2023-04-02:
Where can i get this esp8266?

RuhtraXD, 2023-04-02:
Plis can you make a more detailed step by step, im thinking im doing something wrong.

gulsinho06, 2023-04-02:
Okay, the tutorial is in portuguese but you can translate https://www.robocore.net/tutoriais/instalando-driver-do-nodemcu
you must install CH340g driver

RuhtraXD, 2023-04-02:
Alright, the first tutorial i already done yesterday and it doens't work (portuguese is not a problem, im brazilian) and now i've this second tutorial, now i can flash the program on the arduino normally?

gulsinho06, 2023-04-02:
Sou br tbm posso tentalo ajudalo meu dc para contato: SPARKINHO#9289

RuhtraXD, 2023-04-02:
mostro aqui que nao da pra enviar o pedido, Ruhtra#6973 tenta o meu

closed 2023-04-05 by RuhtraXD