function3d/petalot #33: end of course question!

gulsinho06, 2023-04-12:
Can I use this one?

https://pt.aliexpress.com/item/1005005327520357.html?spm=a2g0o.cart.0.0.28a27f06D8TvWH&mp=1&gatewayAdapt=glo2bra

function3d, 2023-04-12:
I don't know the dimensions of that one but if it's not the same it won't
fit.

gulsinho06, 2023-04-12:
Where can I find the IRLZ44N the link you provided is no longer available

function3d, 2023-04-12:
https://a.aliexpress.com/_EQTfu8r

https://a.aliexpress.com/_EuAvhxp

closed 2023-04-17 by gulsinho06

HumbleDeer, 2023-05-01:
> Where can I find the IRLZ44N the link you provided is no longer available

You can use another suitable N-channel MOSFET if you do not have this specific one on hand, as long as the pinout matches and the specifications are roughly enough for the project (eg voltage across it, and the current drawn by the heater)