function3d/petalot #39: Include ESPHome config file

HumbleDeer, 2023-05-01:
The ESPHome config file is required to modify the firmware. Included in the repository is only the files that ESPHome generated using the PlatformIO environment, but this is not enough to recreate the ESPHome project if any adjustments to firmware are required.

The original ESPHome config file comes in the form of a .yaml file.

HumbleDeer, 2023-05-03:
Oh, how come the use of ESPHome as a platform then?

function3d, 2023-05-03:
I don't know what ESPHome is, I just use the Arduino IDE.

HumbleDeer, 2023-05-03:
It's mentioned on the readme for the firmware as a requirement. 😅
[image]
If you use just Arduino, I will look into making some small improvements to the code to make it C++ standard compliant (proper usage of header files etc)

function3d, 2023-05-03:
Esphome-flasher is a software only for loading firmware compiled by Arduino
IDE

HumbleDeer, 2023-05-03:
I see.
I'll do some work on the firmware then, and plan on adding some optional
features I'd like to have. I'll do that on my fork. Cheers.

closed 2023-05-09 by function3d