function3d/petalot #42: No matter what i do, the heater block doesn't heat.

RuhtraXD, 2023-05-25:
i already have all the peaces of petalot, but, for some reason the heater block just dont heat. I tested all the parts that could have a bad function or problem, and everything works properlly, but when i combine everything, doesn't heat. what could be the problem?

function3d, 2023-05-28:
I can only think of a bad weld, check that there are not two adjacent welds communicating or similar

JesConsuegra, 2023-06-23:
I had the same problem. At the end it was caused by a faulty MOSFET. Ensure it is able to support the load. To test it unplug the heating element and check if there are 12 volts between the two poles of the plug in the board when you activate it from the petalot web page. If you get 12 volts , try to connect again the heating element in series with an ampmeter. If there is not enough current flowing, replace the MOSFET.

RuhtraXD, 2023-07-24:
OH i forgot to close the issue. I solve the problem a couple weeks ago and the problem was that the guy who soldered the pcb damaged the board in the mosfet GND conection and it make the heat block unable to heat, i solve using another new pcb, thank you for the support guys.

closed 2023-07-24 by RuhtraXD