function3d/petalot #43: Stepper motor problem

eteache, 2023-06-02:
Hi,

I've assembled everything, but the motor doesn't turn. What checks can I do? 
Is it normal for the motor driver to get so hot even with the heat sink?

function3d, 2023-06-04:
Hi,  you probably have a short circuit, check the adjoining welds that are
not communicating.

scalinux, 2023-06-04:
Try testing the motor without drive using this as example. Also check youtube videos explain how to. Good luck!

[image]

eteache, 2023-07-14:
Motor is fine, tried to connect directly to my 3D printer and works fine. After two boards built, I think I'm welding bad some component...

closed 2024-01-16 by function3d