function3d/petalot #44: Arduino IDE

gulsinho06, 2023-06-09:
For which board should I compile the code? inside the esp8266 library

function3d, 2023-06-13:
[image]

function3d, 2023-06-13:
this is my complete list of libraries, it doesn't mean that all of them are necessary to compile the firmware, but they are all the ones that I currently have and the firmware compiles for me:

[image]
[image]
[image]
[image]
[image]
[image]

closed 2023-07-22 by gulsinho06