function3d/petalot #49: ABOUT ESP8266MOD Flash

scalinux, 2023-07-11:
After flash the module it get into a loop reboot. 

Re flash with original dump firmware from manufacture and get back to normal. ESP8266MOD USB-C type. [image]

function3d, 2023-07-11:
This is not the module for which the firmware is compiled

scalinux, 2023-07-12:
Thanks.
After testing and testing keep trying the .bin file. Manage to make it work without using the flash ESPTool recommended on the firmware article. 

The only change in this module is USB-C. The only way get it to flashes is via web interface ESP Web Tool, using MacBook Pro M2.

First need to flash with official ESP HOME. Second step set WIFI. Then using the https://web.esphome.io click Connect then Install the PETALOT firmware.

Make sure connect 3V on the pins. Then the USB-C to flash.

[image]

Then flash Official ESPHome.
[image]

Or web Tool.
[image]

Then as usual test the local ip web interface. 

[image]

Your.bin file loaded in the Mini D1 USB-C 

Thanks for sharing this project.
#scalinux

function3d, 2023-07-12:
Thank you very much for that exhaustive documentation of the case

scalinux, 2023-07-12:
Makes sure to add this to the article for GNU/Linux/MAC users.

Thanks..

function3d, 2023-07-12:
I am a Linux user and the steps I describe work well for me.

closed 2023-07-13 by scalinux