function3d/petalot #59: Dificultad para cortar botellas.

dren-lab, 2024-01-17:
Hola, gracias por el proyecto, los vídeos y por ser de ayuda.
Todavía no he descubierto qué estoy haciendo mal pero no puedo cortar las botellas.
Había invertido los rodamientos, donde el rodamiento más alto estaba en el lado donde descansa la garra. Lo arreglé.
Sin embargo, cuando empiezo a tirar de la tira, la botella "sube" y se sale de la "guía", y la tira se estrecha y se rompe. O la botella sigue moviéndose hacia arriba y hacia abajo por la guía y la tira no tiene un tamaño uniforme.

ola, obrigado pelo projeto, pelos videos e por ser solicito.
eu ainda nao descobri o que eu estou fazendo de errado mas eu nao estou conseguindo cortar as garrafas. 
Eu tinha invertido os rolamentos, onde o rolamento mais alto ficava do lado onde se apoia a garrava. consertei isto.
Porem quando eu começo a puxar a tira, a garrafa "sobe" e sai do "guia", e a tira se estreita e parte. Ou a garrafa fica subindo e descendo do guia e a tira nao fica com tamanho uniforme.
[image]
[image]
[image]
[image]
[image]
[image]

JesConsuegra, 2024-01-20:
Hola Samuel. Yo he montado varios (gracias @function3d  por el diseño) y por lo que veo en tus fotos el corte no es limpio, el afilado de los bordes del rodamiento no debe estar en condiciones. Cuando corta bien va muy suave y queda un corte muy limpio, sin rebabas y suave al tacto. Por otra parte hace falta algún peso para mantener la garrafa en contacto con la base, yo uso una bobina vacía de las de PLA puesta encima del cuello de la garrafa, y con eso va bien. Depende del peso de la bobina puede que haya que poner dos en vez de una. Prueba de "afilar" el rodamiento pasándolo por encima de una hoja de papel de lija sobre una madera plana (con plancha de melamina encima, por ejemplo).

dren-lab, 2024-01-23:
Hola, gracias por responder. Gracias por el consejo de intentar lijar el rodamiento en una superficie plana, porque al lijar con la amoladora, la rectificadora, me da miedo que el rodamiento quede con el borde torcido o inclinado. Adjunto una imagen que es mejor que tratar de explicar con palabras.

[image]

ola, obrigado por responder. agradeço pela dica de tentar lixar o rolamento em uma superficie plana, pois lixando com o esmeril, a maquina esmerilhadeira, eu tenho receio de que o rolamento fique com a borda torta, ou inclinada. estou anexando uma imagem que é melhor do que tentar explicar com as palavras.

function3d, 2024-01-23:
En realidad es preferible lo primero a lo segundo, nunca lo tercero (orden:
izquierda-derecha, arriba-abajo), me refiero a la imagen que has adjuntado,
es preferible que esté ligeramente no horizontal inclinado hacia el
interior del rodamiento, funciona igual o mejor que completamente
horizontal ya que horizontal perfecto es complicado de conseguir

function3d, 2024-01-23:
Tutorial actualizado de como hago el cortador de botellas, incluyo STLs en la descripción compatibles con vuestros actuales rodamientos tornillos y guia https://youtu.be/hEdDhRgakms?si=jsJoxt1nZHrgvoxq

dren-lab, 2024-01-24:
este seria o "bottle-cutter-2.0" ?? preferio fazer só com o 608zz?

function3d, 2024-01-24:
De la versión inicial, desde hace tiempo, hay una versión para 608zz, por
tus fotos no tengo claro cuales son tus rodamientos pero yo llevo haciendo
cutters con el 608zz más de un año, no es que sean preferibles es que son
más fáciles de encontrar y económicos.

closed 2024-02-17 by function3d