function3d/petalot #60: Dificultad con una de las piezas del nuevo cortador de botellas

JesConsuegra, 2024-01-27:
Hola, en primer lugar felicitarte por el excelente rediseño del cortador, he visto el video y me ha parecido una idea estupenda, por lo que me he lanzado a convertir los mios al nuevo diseño.
La pieza base se ha impreso sin ningún problema, pero la pieza 2 (la ruleta que sujeta el tornillo) al ir a laminarla, Cura 5.6 me dice que la pieza es diminuta y la amplia automñaticamente por 100.000 con lo que queda una pieza enorme que llena la base de impresión. Le he quitado un cero y con ampliación de 10.000 aparentemente sale de un tamaño correcto, se imprime correctamente pero al ponerla en la base queda un poco suelta y al meterle el tornillo sólo entra las primeras dos vueltas.
Podrías por favor revisar a ver si ha habido algñun problema al subir la pieza?
La he cogido de Printables.
Muchas gracias!
Jes.

JesConsuegra, 2024-01-27:
Probado tambien con IdeaMaker 4.3.3 con el mismo resultado.
Saludos,
Jes.

function3d, 2024-01-27:
A mí me dice algo parecido en orca slicer pero el mensaje es que parece
estar en pulgadas y que sí quiero convertirlo a milímetros y al decirle que
si me lo deja bien, tengo que investigarlo

function3d, 2024-01-29:
Solucionado, actualizado en printables

closed 2024-01-29 by function3d

JesConsuegra, 2024-01-29:
Mil gracias!!